sblibs/pySwitchbot #9: Pi zero w running homebridge

mrbillpapas, 2020-02-19:
Hi! this is awesome you were able to get the password working! Right now I am using a pi zero w running home bridge and script2 to execute the following command: "gatttool -t random -b XX:XX:XX:XX:XX:XX --char-write-req -a 0x0016 -n 570100" it works but I would really love to set a password for the swtichbot! I am a complete newbie when it comes to python and not sure how to integrate this library into my setup, or if it would even work on the pi zero. Do you by any chance have any pointers? Could i configure gatttool to send the correct command to a password protected switchbot? Thank you so much!

moretea, 2020-12-29:
I suppose you could use the SwitchBot app to connect to the SwitchBot device, and set a password?

bdraco, 2022-08-28:
This doesn't look like an issue about this library.

closed 2022-08-28 by bdraco