sblibs/pySwitchbot #11: Add Switchbot Curtain support

dshanske, 2020-11-15:
https://github.com/OpenWonderLabs/python-host/commit/926a0990a2156da60a8c0f73423d7fd192769b38

Looks like the coding is in the manufacturers repo.

Danielhiversen, 2020-11-15:
I do not have the Switchbot Curtain, so I am not going to add support for it.

dshanske, 2020-11-15:
@Danielhiversen Understandable. How would you feel about a PR? I'm not much of a Python programmer, but I think I could handle this. I want to try to upstream it into Home Assistant, where I believe you are also the maintainer of the component.

Danielhiversen, 2020-11-15:
A pr would be great 👍

daenny, 2020-12-18:
https://github.com/Danielhiversen/pySwitchbot/pull/13 initial work is done here.

moretea, 2020-12-19:
#13 is merged, so AFAIK this ticket can be closed.

closed 2020-12-19 by Danielhiversen