sblibs/pySwitchbot #12: Having trouble using this from Pi3

mplacona, 2020-12-02:
Heya, first of all, thanks for the library. I am trying to use this library through the [switchbot_mqtt](https://github.com/fphammerle/switchbot-mqtt) library but when it gets to it using this library everything just hangs and nothing happens.

I chatted with the developer from switchbot-mqtt and [after some debugging](https://github.com/fphammerle/switchbot-mqtt/issues/19), he suggested I try to ask it here. 

I'm at the very point where even if I try to do:

```
import switchbot
switchbot_device = switchbot.Switchbot(mac="aa:bb:cc:dd:ee:ff")
switchbot_device.turn_on():
```

The whole thing just hangs and nothing happens.

I'd appreciate some help here and obviously am willing help debugging.

Thank you in advance

fphammerle, 2020-12-02:
I assume that either of these two lines fails:
```python
        hand_service = self._device.getServiceByUUID(UUID)
        hand = hand_service.getCharacteristics(HANDLE)[0]
```
(https://github.com/Danielhiversen/pySwitchbot/blob/0.8.0/switchbot/__init__.py#L75)
as @mplacona gets the following log output:
```
received topic=homeassistant/switch/switchbot/{{MAC}}/set payload=b'ON'
Sending command to switchbot 570101
Connecting to Switchbot...
Connected to Switchbot.
Prepare to send
```
missing the output of `_LOGGER.debug("Sending command, %s", key)`

fphammerle, 2020-12-03:
@mplacona, could you check if the following script works on your RPI3 ?

```python
import bluepy

UUID = "cba20d00-224d-11e6-9fb8-0002a5d5c51b"
HANDLE = "cba20002-224d-11e6-9fb8-0002a5d5c51b"

device = bluepy.btle.Peripheral("aa:bb:cc:dd:ee:ff", bluepy.btle.ADDR_TYPE_RANDOM)      

hand_service = device.getServiceByUUID(UUID)      
print(hand_service)

hand = hand_service.getCharacteristics(HANDLE)[0]
print(hand)
```
(relevant lines of pySwitchbot)

you should get:
```
Service <uuid=cba20d00-224d-11e6-9fb8-0002a5d5c51b handleStart=[...]>
Characteristic <cba20002-224d-11e6-9fb8-0002a5d5c51b>
```

mplacona, 2020-12-03:
Heya, just tried this one real quick and it hangs the same way.

Also nothing on `journalctl -f` 😞

fphammerle, 2020-12-03:
Thanks! Probably hangs somewhere around that line then: https://github.com/IanHarvey/bluepy/blob/v/1.3.0/bluepy/btle.py#L487

bdraco, 2022-08-28:
This library no longer uses bluepy

closed 2022-08-28 by bdraco