sblibs/pySwitchbot #17: Feature request: Add support for SwitchBot Meter

moretea, 2020-12-29:
Add support for the [SwitchBot Meter](https://www.switch-bot.com/products/switchbot-meter).

There already exists a Python script that scans BLE advertisements and publishes the results to MQTT (https://github.com/ronschaeffer/sbm2mqtt).

RenierM26, 2021-06-08:
Hi @moretea,

Any chance you could send the met the Service Data for a Meter?

1) You can use the nRF Connect android app (available via play store)
2) Open app and expand your device. The last line "services data" is what I need.
3) Would you mind changing the temp between F and C and capture the data for both? From the code examples it looks like the unit of measurement is part of the services data. 

Just for interest, the services data is a hex code with each byte representing some or other metric returned in these devices.

RenierM26, 2021-06-12:
#23

closed 2022-08-28 by bdraco