sblibs/pySwitchbot #18: Feature request: Add support for reading the state from Switchbot devices

moretea, 2020-12-29:
Based on the observed behavior in the Switchbot app, it must be possible to query for the current state from the Switchbot devices. 

I'm not intimately familiar (yet) with BLE, but from what I gather, we could fetch these as characteristics. This poses the question of the impact on battery life if we poll to often.

basvdl, 2021-02-18:
I have succeeded reading the battery, position and light level using the python package bleak. The `bluepy` package is not working for me, it does not fetch any advertisement data regarding battery, position or light level. I'm new to bluetooth communications, so I might have done something wrong using `bluepy`. I also like bleak more, since it works cross platform, whereis `bluepy` only works on Linux.

closed 2021-04-15 by Danielhiversen

oetelaar24, 2021-05-12:
Hi, I'm really keen on trying this additional functionality out. Any chance you are willing to make a new release containing it?

Danielhiversen, 2021-05-12:
PySwitchbot 0.10.0 is now released

PR #21 (2021-04-15): feature: reading states from curtain
Implemented support for reading battery percent, position and light level from curtain.
Added reverse attribute for curtain. Useful when using a silder to set position.

Fixes #10 
Fixes #18