sblibs/pySwitchbot #31: Battery sometimes errnoeously reports `0`

bachya, 2021-10-29:
Hi! Reporting here at the request of the owner of [`switchbot-mqtt`](https://github.com/fphammerle/switchbot-mqtt) (specifically: https://github.com/fphammerle/switchbot-mqtt/issues/55)

I have an automation that gets new battery info every hour. I'm noticing that every so often, the is reported as `0` for a time. Sometime after, it returns to a "normal" level.

Here's a history graph from Home Assistant:
[image]

Notice that the battery values shows `0` at the top of the hour (right when my automation would have run).

Unfortunately, `switchbot-mqtt`'s debug-level logs don't show any smoking gun – in the latest instance of `0` being reported, all I see from that library are:

```
Oct 29 11:51:53 pmc-vm-switchbot-mqtt.phil.servers switchbot-mqtt[6666]: 2021-10-29T11:51:53-0600:DEBUG:switchbot_mqtt._actors._base:_mqtt_update_device_info_callback:received topic=homeassistant/switch/switchbot/fd:64:1b:3e:ce:bf/request-device-info payload=b''
Oct 29 11:51:58 pmc-vm-switchbot-mqtt.phil.servers switchbot-mqtt[6666]: 2021-10-29T11:51:58-0600:DEBUG:switchbot_mqtt._actors._base:_mqtt_publish:publishing topic=homeassistant/switch/switchbot/fd:64:1b:3e:ce:bf/battery-percentage payload=b'0'
```

Lastly, in case it's useful, there are times where my BLE antenna fails to connect to my Switchbot (to be expected, since they're about 30 ft. apart). I don't see a direct correlation between connection issues in the logs and this issue, but want to mention it just in case?

Any thoughts on something in pySwitchbot that might be causing this? Thanks!

RenierM26, 2021-10-30:
Hi bachya,

There are a lot of changes in the latest pyswitchbot version. Perhaps try the latest one? The old version had bit error which most likely caused the 0 return. (BTLE data is a binary tree with a lot of intertwined data)

bachya, 2021-10-30:
Thanks, @RenierM26! `switchbot-mqtt` uses pySwitchbot >=0.10.0,<0.11 – I'll upgrade to 0.12.0, see what happens, and report back.

bachya, 2021-10-30:
After several hours, 0.12.0 is looking good! I'll close this out and follow up on the other issue. Thank you!

bachya, 2021-10-31:
Unfortunately, even after upgrading to 0.12.0, this is still occurring (although it's far less often). Please let me know what other info I can provide!

bdraco, 2022-08-28:
@bachya Is this still an issue?

bachya, 2022-08-28:
Haven't seen this since I started using the HASS Bluetooth layer.

closed 2022-08-28 by bdraco