sblibs/pySwitchbot #37: how to use from terminal?

zen85, 2022-02-04:
Hello dear Developers,

I managed to get this project to run in node-red thanks to the documentation:
https://github.com/OpenWonderLabs/python-host
with a command in an exec node like:
`sudo python3 python-host/switchbot_py3.py -d XX:XX:XX:XX:XX:XX -c on`

but i am missing the possibility to get the batterylevel very much and therefor i found your library. I just dont know how to use your library in nodered.

Is it possible to write some kind of documentation to get me started?
i did: 
 git clone https://github.com/Danielhiversen/pySwitchbot.git
 
 so what next?
 
 sorry if this question is stupid or to newbish but i would really like to use this and i just dont know what to do next :(

ninadpchaudhari, 2022-06-06:
@zen85 It's been long, let me know if you still have issues.

bdraco, 2022-08-28:
Answered above in https://github.com/Danielhiversen/pySwitchbot/issues/41#issuecomment-1147933540

closed 2022-08-28 by bdraco