sblibs/pySwitchbot #108: SwitchBot Plug Mini (JP) Model No. W2001400

SeraphicRav, 2022-09-11:
Hello !

I am Home Assistant user and while troubleshooting my installation, I discovered that Home Assistant was relying on this project. Good job supporting meters and bots !

I am in Japan and the mini plugs are not detected and I was wondering how I could add them.

I have a little experience in Python but no experience with Bluetooth protocol, the README does not help me much to understand what do either lol

So what can I do ?

Best regards

bdraco, 2022-09-11:
This one probably has the same protocol but a different device char.

Can you turn on logs for `homeassistant.components.bluetooth` and send a copy of the advertisement data for the device?

SeraphicRav, 2022-09-12:
In the Switchbot app, it does not show the Bluetooth mac address for the plugs (and the BLE mac address paired curtains if they are even visible ?).

Here are the logs for devices I could not identify, sorry, I don't how to interpret these data:

```text
hci0: XX:XX:XX:XX:XX:45 AdvertisementData(manufacturer_data={89: b'\xe6\x8c\x16yjE'}, service_data={'00000d00-0000-1000-8000-00805f9b34fb': b'T\x00A\x08\x9c5'}, service_uuids=['cba20d00-224d-11e6-9fb8-0002a5d5c51b']) connectable: True match: {'switchbot'} rssi: -83
hci0: XX:XX:XX:XX::78 AdvertisementData(manufacturer_data={2409: b'\xef\xc9C\xd3\xedx\xc9\td1\x03'}, service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': b'c@1d1\x03'}) connectable: True match: {'switchbot'} rssi: -66
hci0: XX:XX:XX:XX::04 AdvertisementData(manufacturer_data={2409: b'\xfb6T\xdc\xa7\x04\xb7\rd\x11\x14'}, service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': b'c@Fd\x11\x14'}) connectable: True match: {'switchbot'} rssi: -52
```

As a bonus, the logs for a remote:

```text
hci0: FC:24:03:3C:FD:3B AdvertisementData(manufacturer_data={89: b'\xfc$\x03<\xfd;'}, service_data={'00000d00-0000-1000-8000-00805f9b34fb': b'b\x00d\x00'}, service_uuids=['cba20d00-224d-11e6-9fb8-0002a5d5c51b']) connectable: True match: {'switchbot'} rssi: -75
```

Are the logs of the plugs included ? I filtered the logs with switchbot and used the commands `sort` and `uniq` on the log output.

bdraco, 2022-09-12:
https://github.com/Danielhiversen/pySwitchbot/blob/9014b8411fe2b1d201ed4e4fe7b07bde52d33134/switchbot/adv_parser.py#L36

This is from a `WoSensorTH` (first char of the service_data is `T`)
hci0: XX:XX:XX:XX:XX:45 AdvertisementData(manufacturer_data={89: b'\xe6\x8c\x16yjE'}, service_data={'00000d00-0000-1000-8000-00805f9b34fb': b'T\x00A\x08\x9c5'}, service_uuids=['cba20d00-224d-11e6-9fb8-0002a5d5c51b']) connectable: True match: {'switchbot'} rssi: -83

These are from a curtain device (first char of the service_data is `c`)
hci0: XX:XX:XX:XX::78 AdvertisementData(manufacturer_data={2409: b'\xef\xc9C\xd3\xedx\xc9\td1\x03'}, service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': b'c@1d1\x03'}) connectable: True match: {'switchbot'} rssi: -66
hci0: XX:XX:XX:XX::04 AdvertisementData(manufacturer_data={2409: b'\xfb6T\xdc\xa7\x04\xb7\rd\x11\x14'}, service_data={'0000fd3d-0000-1000-8000-00805f9b34fb': b'c@Fd\x11\x14'}) connectable: True match: {'switchbot'} rssi: -52

Can you check the mac address of the plug in the app and then find the advertisements based on it?

SeraphicRav, 2022-09-12:
Oh I see, so it is my paired curtains and maybe a neighbor has a thermometer.

Sorry, I have been checking on internet why I did not have a BLE MAC address: I have an old version of the plug which actually do not have Bluetooth.

closed 2022-09-12 by SeraphicRav