sblibs/pySwitchbot #136: Position feedback not reported correctly since upgrading to v6.0

georgesak, 2022-11-12:
Upgraded my curtains to firmware V6.0 and since then, I don't get proper feedback. Curtains always show as open. Both control and battery feedback are still working however.

This is using Home Assistant.

bdraco, 2022-11-12:
Please post the new AdvertisementData from turning on homeassistant.components.bluetooth

georgesak, 2022-11-13:
Is that what you're looking for? First one is when they were open, 2nd one when closed. Note that it's a pair of curtains that are tied together.

AdvertisementData(local_name='WoCurtain', manufacturer_data={89: b'\xf5\x98\x94\x08\xa0\xe7', 2409: b'\xf5\x98\x94\x08\xa0\xe7\x9b\x0f\x00"\x04'}, service_data={'00000d00-0000-1000-8000-00805f9b34fb': b'c\xd0H\x00\x12\x04', '0000fd3d-0000-1000-8000-00805f9b34fb': b'c\xc0G\x00"\x04'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', 'cba20d00-224d-11e6-9fb8-0002a5d5c51b'], rssi=-62) connectable: True match: {'switchbot'} rssi: -62

AdvertisementData(local_name='WoCurtain', manufacturer_data={89: b'\xf5\x98\x94\x08\xa0\xe7', 2409: b'\xf5\x98\x94\x08\xa0\xe7\xa5\x0fc"\x04'}, service_data={'00000d00-0000-1000-8000-00805f9b34fb': b'c\xd0H\x00\x12\x04', '0000fd3d-0000-1000-8000-00805f9b34fb': b'c\xc0Gc"\x04'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', 'cba20d00-224d-11e6-9fb8-0002a5d5c51b'], rssi=-67) connectable: True match: set() rssi: -67

closed 2022-11-13 by bdraco

bdraco, 2022-11-13:
I upgraded my test one to firmware 6 and fixed it reading the wrong service data.

Hopefully thats the issue.  I don't have a second one here so I couldn't verify that

bdraco, 2022-11-14:
I picked up a second one this morning to make sure I adjusted for the firmware change and its working as expected when linked

PR #138 (2022-11-13): Add tests for firmware v6
closes #136
PR #139 (2022-11-13): Fixes for curtain firmware 6
fixes #136
PR #140 (2022-11-14): Fix v6 firmware with bluez
The previous fix was tested on macos which
does not have the update history behavior

fixes #136