sblibs/pySwitchbot #166: Add WoCeilingPro

jurmarcus, 2022-12-25:
Looks like WoCeilingPro should be added as a new device here. I have both he Add WoCeiling and WoCeilingPro Lights, and I could test this and add it but I am not sure what is the quickest way to get this up and running and I am using this plugin through home assistant VM.

https://github.com/Danielhiversen/pySwitchbot/blob/a804c7863b253dfe0b09bdca3cd724d21c201492/switchbot/const.py#L24

bdraco, 2022-12-25:
I don't have a Pro, but if you only a PR to add support for it, I'm happy to review it

jurmarcus, 2023-01-08:
so how that the new year has started, what is the best way to get a test env up and running for pyswitchbot? is it through some type of ha-core setup?

bdraco, 2023-01-08:
https://developers.home-assistant.io/docs/development_index

closed 2024-01-03 by bdraco