sblibs/pySwitchbot #172: Adding support for new SwitchBot Blind Tilt

PaleoW, 2022-12-30:
Hi,

I just received my [SwitchBot Blind Tilt](https://www.kickstarter.com/projects/switchbot/switchbot-blind-tilt-solar-powered-retrofit-smart-blinds) , and I would like to see if I can be of any assistance to help adding support to pySwitchbot. For example,  is there any bluetooth capture I could provide to help more capable people implement the protocol for these new devices ?

Regards

warrenrees, 2023-01-06:
I can assist with this. I'll try to spend some time tonight to write a quick example program so I can extend the library to support Blind Tilt and the Hub mini (gets detected as plug mini for some odd reason )

warrenrees, 2023-01-07:
Thank you @jesserockz you beat me to the punch and did a much cleaner job than my hack job i was doing.  Tested the code in HA after tweaking the switchbot component and it appears to work great:

[image]

donavanbecker, 2023-01-07:
@jesserocks could you update the SwitchBotAPI-BLE repo with your finding.

jeremywgleeson, 2023-01-22:
@jesserockz
@BTMorton

Because HA core repo's policy on integration feature requests is to have them posted on the forum, would it be possible to post your proposed core implementation behavior (mentioned at https://github.com/Danielhiversen/pySwitchbot/pull/181#issuecomment-1399149488) on the feature request to get community feedback? (Here's the thread I found requesting support for the blind tilt: https://community.home-assistant.io/t/support-for-switchbot-blind-tilt/518207)

I'd do it myself but am unable to find Jesse's branch on core to fully understand the proposed behavior.

jesserockz, 2023-01-25:
@jeremywgleeson My code does not require changes to HA cores structure. It just exposes a cover entity with only tilt control, no open/close control which is already supported with `supported_features`

PR #181 (2023-01-25): Initial Blind Tilt support
closes #172
closed 2023-01-25 by bdraco