sblibs/pySwitchbot #179: [SwitchBot Lock Encryption Key] Getting Failed to Authenticate

weitheng, 2023-01-05:
Hi I am getting `Failed to Authenticate` when trying to extract the SwitchBot Lock encryption key. My SwitchBot account is created and logged in using Apple 3rd Party Account, could that be the reason, if so, is there any fix to allow extraction of encryption key using 3rd party account?

bdraco, 2023-01-05:
The tools can only use switchbot accounts

christi-g, 2023-01-05:
I have the same problem. I use the "sign in with apple" in the Switchbot app... 
Somebody knows how to convert the account in "normal" Switchbot account?

closed 2023-01-05 by weitheng

KiraPC, 2023-01-20:
Is there no way to enable 3rd Party Account? I'm logged in with google but I need encryption keys.

captshadab, 2023-02-24:
this is so inconvenient. does switchbot wants us to transfer all devices to new account. this is absolutely insane. i have 20 devices around