sblibs/pySwitchbot #186: Support for Lock v6.4 - Door without outside handle

KiraPC, 2023-01-24:
I have the last firmware installed on Lock v6.4 that enabled support with door without external handle. 

This let the door to keep open the lock in the unlock status. After this, the lock change its state in locked but for the component it is in unlocked state, so, when I try to open again the door, nothing happen.

Is there a way to change its behavior?

Looking at the source code of the pySwitchBot library I think I found something.

Stay at my last logs, the status of the lock should be in 'LockStatus.LOCKING'.

In the file https://github.com/Danielhiversen/pySwitchbot/blob/master/switchbot/devices/lock.py#L163 a sets {LockStatus.LOCKED, LockStatus.LOCKING} is sent to the function _lock_unlock (https://github.com/Danielhiversen/pySwitchbot/blob/master/switchbot/devices/lock.py#L179) as ignore_statuses parameter.

So if the lock is in this status, the module does not send the unlock command.

Discussing also in:

KiraPC, 2023-01-30:
@Danielhiversen I'm trying to implement this feature by my self and then open a PR.

What I would like to achieve is to add a new function to the Lock device to retract the deadbolt.

I have now a question, where I could find the command to set the Lock to this status? I'm not confident with BLE.

vinz486, 2023-06-26:
Hi,
I'm also very interested (all EU users should). I have a Lock device and some Hackrf tools. Can I try to detect BLE commands.

KiraPC, 2023-07-05:
> Hi,
> 
> I'm also very interested (all EU users should). I have a Lock device and some Hackrf tools. Can I try to detect BLE commands.

Have you detected the command?

vinz486, 2023-07-06:
> > Hi,
> > I'm also very interested (all EU users should). I have a Lock device and some Hackrf tools. Can I try to detect BLE commands.
> 
> Have you detected the command?

Unfortunately no: I cannot detect ble packets, I don't know why.

Michal4K, 2023-08-11:
I am also really interested in this feature (bought this lock few weeks ago). I think for basic/unproper implementation its not even needed to make another another protocol reverse engineering attempt. Because EU type lock react on unlock command send by this library also with unlatching. So, if you don't need feature to just unlock door without unlatching it, just adding another function to send unlock command even when the lock reports unlock state should be enough. Than you just add switch to HS integration to trigger this function.

I think, i have some nRF board, that i could use to sniff ble packet, so maybe i should give a try to implement this even with all states.

Michal4K, 2023-08-14:
I was able to capture and decrypt command to just unlock without unlatching.
`COMMAND_NOT_FULLY_UNLOCK = f"{COMMAND_HEADER}0f4e010110a0 "`

Now i consider how this should be properly implemented.

KiraPC, 2023-08-14:
Amazing. I did some code study time ago. Ill check in I have time

Michal4K, 2023-08-16:
I implemented in this [PR](https://github.com/Danielhiversen/pySwitchbot/pull/215).

Michal4K, 2023-08-30:
@KiraPC I think you can close this issue now. Support for this is already implemented and also it's merged in HA https://github.com/home-assistant/core/pull/98620

KiraPC, 2023-08-30:
> @KiraPC I think you can close this issue now. Support for this is already implemented and also it's merged in HA https://github.com/home-assistant/core/pull/98620 

In which version of HA will be delivered?

closed 2023-08-30 by KiraPC

Michal4K, 2023-08-30:
> > @KiraPC I think you can close this issue now. Support for this is already implemented and also it's merged in HA [home-assistant/core#98620](https://github.com/home-assistant/core/pull/98620)
> 
> In which version of HA will be delivered?

I don't know probably in the next one - it was my first contribution to HA. I am not sure how they deploy releases.

bdraco, 2023-08-30:
In beta today. Release next week