sblibs/pySwitchbot #191: transfer apple account to normal account

captshadab, 2023-02-24:
hi

what is the issue with switchbot. I have 20 devices and i can't get encryption key for my lock because year ago I sued apple id. now no one can help me. why make a system which would create issues for customers rather than helping them. why can't you convert my apple account to normal account. so, in short I am stuck and i can never use lock with home assistant ever. very pathetic

asknoone, 2023-02-27:
Yeah, I have same issue as I created my Switchbot Account with Google SSO.

Hopefully someone will come up with a solution soon as I would love to use Bluetooth proxy instead of Switchbot's unreliable cloud-based API.

I'm guessing we can probably create a new Switchbot account, reset the lock and assign it to the new account; but feels like a hassle.

captshadab, 2023-02-27:
> Yeah, I have same issue as I created my Switchbot Account with Google SSO.
>
> Hopefully someone will come up with a solution soon as I would love to use Bluetooth proxy instead of Switchbot's [...]

Hey I have a good news. I emailed switchbot about the issue and they mentioned that they have a way by which they can transfer our account to non third party linked

After that we are free to have lock encryption key via any method.

And absolutely even i plan to use Bluetooth proxy for all my bots locks curtains. Hopefully they respond soon as it's been 2 days since last reply from them 

I urge even you should email them