sblibs/pySwitchbot #212: I get ModuleNotFoundError

taikun114, 2023-06-10:
Hello.

I used this script to try to get the encryption key to add a SwitchBot Lock to my Home Assistant.
However, when I run the script I get the following error.

```
Traceback (most recent call last):
  File "/Users/taikun/pySwitchbot-0.37.6/scripts/get_encryption_key.py", line 5, in <module>
    from switchbot import SwitchbotLock
ModuleNotFoundError: No module named 'switchbot'
```

What should I do?

I have run this script from the macOS terminal and from the Advanced SSH & Terminal in the Home Assistant, and both show the exact same error.

By the way, the method to get the encryption key using the username and password of the SwitchBot account in the Home Assistant did not work. That is why I decided to use this script.

Thank you.

buss0140, 2023-06-11:
I am having the same issue.

Yogsther, 2023-11-14:
Moving the `switchbot` folder into `scripts` worked for me.

taikun114, 2023-11-15:
> Moving the `switchbot` folder into `scripts` worked for me.

Thank you for the information!
I tried it but unfortunately it did not work with the following error message.

```
Traceback (most recent call last):
  File "/Users/taikun/pySwitchbot/scripts/get_encryption_key.py", line 5, in <module>
    from switchbot import SwitchbotLock
  File "/Users/taikun/pySwitchbot/scripts/switchbot/__init__.py", line 4, in <module>
    from bleak_retry_connector import close_stale_connections, get_device
ModuleNotFoundError: No module named 'bleak_retry_connector'
```
[image]

I also tried it with the latest version of pySwitchbot just to be sure, but it showed the same error.

Could you tell me about your environment and what else you did (If you did) when it worked?
I would like to try to reproduce a similar environment as best I can and see if it works.

taikun114, 2023-11-15:
After a little research, I realized that I had not run "setup.py".
I didn't know this because I had no knowledge of Python, but I thought running it might work, so I tried it, but again, it didn't work.

The first thing I tried was to run the following command in the "pySwitchbot" directory.

```
python3 setup.py install
```

This command did not work from the macOS terminal, so I used the Home Assistant terminal.
This command worked without any problems.

I then went to the "scripts" folder and ran the command to get the encryption key, and got the following error.

```
➜  scripts git:(master) python3 get_encryption_key.py XX:XX:XX:XX:XX:XX switchbotaccount@example.com
Traceback (most recent call last):
  File "/homeassistant/pySwitchbot/scripts/get_encryption_key.py", line 5, in <module>
    from switchbot import SwitchbotLock
  File "/usr/lib/python3.11/site-packages/PySwitchbot-0.41.0-py3.11.egg/switchbot/__init__.py", line 22, in <module>
  File "/usr/lib/python3.11/site-packages/PySwitchbot-0.41.0-py3.11.egg/switchbot/devices/lock.py", line 12, in <module>
  File "/usr/lib/python3.11/site-packages/boto3-1.29.0-py3.11.egg/boto3/__init__.py", line 17, in <module>
    from boto3.session import Session
  File "/usr/lib/python3.11/site-packages/boto3-1.29.0-py3.11.egg/boto3/session.py", line 17, in <module>
    import botocore.session
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/session.py", line 26, in <module>
    import botocore.client
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/client.py", line 15, in <module>
    from botocore import waiter, xform_name
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/waiter.py", line 18, in <module>
    from botocore.docs.docstring import WaiterDocstring
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/__init__.py", line 15, in <module>
    from botocore.docs.service import ServiceDocumenter
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/service.py", line 13, in <module>
    from botocore.docs.bcdoc.restdoc import DocumentStructure
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/bcdoc/restdoc.py", line 17, in <module>
    from botocore.compat import OrderedDict
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/compat.py", line 31, in <module>
    from dateutil.tz import tzlocal
ModuleNotFoundError: No module named 'dateutil'
```

I moved the "switchbot" folder to the "scripts" folder and ran the same command again and got the following error.

```
➜  scripts git:(master) python3 get_encryption_key.py XX:XX:XX:XX:XX:XX switchbotaccount@example.com
Traceback (most recent call last):
  File "/homeassistant/pySwitchbot/scripts/get_encryption_key.py", line 5, in <module>
    from switchbot import SwitchbotLock
  File "/homeassistant/pySwitchbot/scripts/switchbot/__init__.py", line 22, in <module>
    from .devices.lock import SwitchbotLock
  File "/homeassistant/pySwitchbot/scripts/switchbot/devices/lock.py", line 12, in <module>
    import boto3
  File "/usr/lib/python3.11/site-packages/boto3-1.29.0-py3.11.egg/boto3/__init__.py", line 17, in <module>
    from boto3.session import Session
  File "/usr/lib/python3.11/site-packages/boto3-1.29.0-py3.11.egg/boto3/session.py", line 17, in <module>
    import botocore.session
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/session.py", line 26, in <module>
    import botocore.client
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/client.py", line 15, in <module>
    from botocore import waiter, xform_name
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/waiter.py", line 18, in <module>
    from botocore.docs.docstring import WaiterDocstring
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/__init__.py", line 15, in <module>
    from botocore.docs.service import ServiceDocumenter
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/service.py", line 13, in <module>
    from botocore.docs.bcdoc.restdoc import DocumentStructure
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/docs/bcdoc/restdoc.py", line 17, in <module>
    from botocore.compat import OrderedDict
  File "/usr/lib/python3.11/site-packages/botocore-1.32.0-py3.11.egg/botocore/compat.py", line 31, in <module>
    from dateutil.tz import tzlocal
ModuleNotFoundError: No module named 'dateutil'
```

It seems that the sentences in the middle or slightly different, but the final error was the same in both cases.

Am I missing something or making a mistake?
If anyone knows, I would appreciate it if you could let me know.

Thank you.

taikun114, 2023-11-16:
After much research and trying, I was able to solve the problem and I would like to note my experience.

I read back the log carefully and noticed the error `error: urllib3 2.1.0 is installed but urllib3<2.1,>=1.25.4 is required by {'botocore'}`, So I uninstalled urllib3, and then installed version 1.25.4, and again tried the `python3 setup.py install` command, but this time I got the following error.

```
error: Setup script exited with error: command 'gcc' failed: No such file or directory
```

I was stuck and couldn't figure out how to install "gcc" on Home Assistant, but then I tried the following command that came to mind.

```
pip install pySwitchbot
```

I was able to run this command with no errors at all.
After running it, I ran the `python3 setup.py install` command again and got a few warnings, but without the errors I had seen earlier.

I then went to the "scripts" folder and ran the command to get the encryption key, which ran without any problems!
I had not moved the "switchbot" folder into the "scripts" folder, but was able to run it without any problems.

I also performed the same procedure on a macOS terminal and was also able to run the command without any problems!

Unfortunately, even after entering the correct email address and password, I got a `Failed to authenticate` error and could not get the encryption key.
This may be because I created my SwitchBot account with a login with Apple ID.

Maybe I made a lot of elementary mistakes because I had no Python knowledge at all, but anyway, the `ModuleNotFoundError` error was successfully resolved, so I will close this issue.
Thank you very much!

closed 2023-11-16 by taikun114