sblibs/pySwitchbot #213: Back to Back commands not executing on WoHand

CDeLeon94, 2023-06-30:
Context: `Home Assistant Core 2023.6.3`  
Which runs `PySwitchbot==0.37.6` for its [Switchbot integration](https://github.com/home-assistant/core/tree/2023.6.3/homeassistant/components/switchbot)  

When I call the 'turn on' service multiple times in a row (using to press a button on a thermostat) the WoHand only executes the press once, despite seeing the `Sending command 5702` the appropriate number of times in the debug logs. 

I believe this is the relevant parts of the logs

```
2023-06-30 08:31:10.937 DEBUG (MainThread) [homeassistant.components.switchbot.entity] Turn Switchbot device on C8:11:60:CA:FF:74
2023-06-30 08:31:10.937 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Scheduling command 570101
2023-06-30 08:31:10.938 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Connecting; RSSI: -46
2023-06-30 08:31:11.142 DEBUG (MainThread) [homeassistant.components.switchbot.entity] Turn Switchbot device on C8:11:60:CA:FF:74
2023-06-30 08:31:11.142 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Scheduling command 570101
2023-06-30 08:31:11.142 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Operation already in progress, waiting for it to complete; RSSI: -46
2023-06-30 08:31:11.856 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Connected; RSSI: -46
2023-06-30 08:31:11.856 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Starting notify and disconnect timer; RSSI: -46
2023-06-30 08:31:11.856 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Subscribe to notifications; RSSI: -46
2023-06-30 08:31:12.305 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Sending command: 570101
2023-06-30 08:31:12.406 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Notification received: 014890
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.bot] WoHand (C8:11:60:CA:FF:74): Turn on result: 014890 -> {'isOn': True}
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Fire callbacks
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Scheduling command 5702
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Already connected before obtaining lock, resetting timer; RSSI: -46
2023-06-30 08:31:12.407 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Sending command: 570101
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Notification received: 014890
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.bot] WoHand (C8:11:60:CA:FF:74): Turn on result: 014890 -> {'isOn': True}
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Fire callbacks
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Scheduling command 5702
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Already connected before obtaining lock, resetting timer; RSSI: -46
2023-06-30 08:31:12.506 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Sending command: 5702
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Notification received: 01643f64000000c40010000000
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Fire callbacks
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Already connected before obtaining lock, resetting timer; RSSI: -46
2023-06-30 08:31:12.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Sending command: 5702
2023-06-30 08:31:12.706 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Notification received: 01643f64000000c40010000000
2023-06-30 08:31:12.706 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Fire callbacks
2023-06-30 08:31:12.706 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:12.706 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Using override value for isOn: True
2023-06-30 08:31:21.108 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Executing timed disconnect after timeout of 8.5
2023-06-30 08:31:21.108 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Executing disconnect
2023-06-30 08:31:21.108 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Executing disconnect with lock
2023-06-30 08:31:21.108 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Disconnecting
2023-06-30 08:31:23.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Disconnected from device; RSSI: -46
2023-06-30 08:31:23.607 DEBUG (MainThread) [switchbot.devices.device] WoHand (C8:11:60:CA:FF:74): Disconnect completed successfully
```

I'm trying to tinker and see what else I can figure out, but the best I can figure out now is that on the WoHand side it's consuming multiple 'Turn_on" commands and processing them as one. My understanding is that after sending the command, the returned `notification` is a *promise* of sorts, and back to back commands receive the same promise, `014890` in the case of the above logs.

bdraco, 2023-08-14:
These device swallow commands sent to quickly. You’ll need to add the delay. 1s seems to be enough

CDeLeon94, 2023-08-14:
Yeah, after further testing it seems like the responses from the device could be used to validate that a command was executed as opposed to ignored. And maybe we could code the library to auto-retry and or auto-delay. I'll try my hand at this if I have some time.