sblibs/pySwitchbot #233: feature request: Switchbot Lock Pro

chrismelba, 2024-04-15:
Hi, 

I've just bought two of the newly released Switchbot Lock Pros. Would it be possible to add them to this repo? Unsure how much they differ from the standard locks.

-Chris

wunderbaum, 2024-05-25:
Would like that, too as I have such a lock.
How can I support that?

ynsgnr, 2024-05-31:
I looked into adding the lock pro and it is discoverable with model `$`:
After making this change you can connect to it by fetching keys with `/scripts/get_encryption_key.py` :

```
    async def get_locks(self) -> dict[str, SwitchBotAdvertisement]:
        """Return all WoLock/Locks devices with services data."""
        locks = await self._get_devices_by_model("o")
        lock_pros = await self._get_devices_by_model("$")
        return {**locks, **lock_pros}
```

To get basic data COMMAND_LOCK_INFO needs to be increased by one from `0f4f8101` to `0f4f8102`. I assume that part of the command is the API Version or model number.

However after applying same byte flips to UNLOCK commands (from 0f4e01011080 to 0f4e02011080) it didn't work. Most unlock commands I tried just worked as lock command with different resources. I tried to brute force it and the logs shows "Locked By X" X being a different service on each command bit increase. Ex: "Unlocked By Google Home"

I'm not familiar with switchbots BLE API though, any help would be appreciated

Michal4K, 2024-05-31:
@ynsgnr
There is no documentation for this commands. The easiest to find right command would be sniff the complication with the android app. You can android developer function Bluetooth HCI snoop for this. I used this technique to find command for opening the door without unlatching on previous model. Unfortunately bluetooth snooping not work on all android phones i was forced to borrow samsung from family member for this, because on Nothing phone 1 the records was always empty.

szclsya, 2024-06-07:
Did some bluetooth snooping and tl;dr: 
+ COMMAND_LOCK_INFO: `0f4f8102`
  - based on @ynsgnr 's work, but this command actually didn't appear in my dump so it might just be a compatibility thingy
+ COMMAND_UNLOCK: `0f4e0101000180`
+ COMMAND_LOCK: `0f4e0101000100`

Tested with my own (North American firmware) Lock Pro.

While we are at it, here's all the commands I found:

```
> 0e01002000008104
> 0f4f8104
> 0f4f0501
> 0f4f0502
> 000501000000006662657a
> 0f4f0401
> 0e01002000008104
> 0f4e0101000100
> 0f4e0101000180
```

varyamereon, 2024-06-07:
Would very much appreciate if the night latch feature would be supported too! This is already a feature of the original switchbot lock.

szclsya, 2024-06-07:
@varyamereon Looks like my model don't have night latch support, so I can't test such feature here. Feel free to snoop it yourself though, I can share the script I use to decrypt the communication.

varyamereon, 2024-06-07:
I think it's supported on all models, though I may be wrong. It's an option during the calibration process. If we don't set it up like that, the door opens every time we unlock which was a bit of a surprise the first time we found the front door wide open 😅

If it's not something you can do I'm happy to help where I can, though I am not familiar with Bluetooth/python at all 🫣

szclsya, 2024-06-07:
Just re-calibrated my lock pro and there were no option for night latch, and since the code implies the original Lock's night latch feature is limited to EU region I'd assume it's indeed still region locked (for some reason).

However since the command for lock/unlock shares a pattern between two generations (i.e lock command both ends in 80), it would be fair to assume unlock with night latch would also use this pattern (so 0f4e010110a0 -> 0f4e01010000a0). I can create a different branch for you to test if you wish.

varyamereon, 2024-06-08:
I can give it a go next week at the earliest. It's not my area of expertise but I should be able to get it working.

Michal4K, 2024-06-08:
It would be also helpful if you can post your lock Advertisement Data from Home Assistant debug log. So we can check if night latch detection works for new model. I assume it will work... but just to be sure

schummar, 2024-06-08:
@szclsya I have the EU lock and logged my bluetooth traffic. `COMMAND_UNLOCK_WITHOUT_UNLATCH` should be 4c53e30a43c214

schummar, 2024-06-08:
@szclsya I have the EU lock and logged my bluetooth traffic. COMMAND_UNLOCK_WITHOUT_UNLATCH should be 0f4e01010001a0

szclsya, 2024-06-08:
@schummar Nice! That proves our theory. I'll add that to the pull request.

varyamereon, 2024-06-09:
Thanks, I appreciate that!

incre77, 2024-06-10:
Thank you very much for your work. Im have been testing but in my case in not fully working.
I made a guest user and i got the encryption key for that user (cis i began with the main account in google third party mode and its imposible to me to reset password or get the encryption key for that account).

I tried this code:
`
async def main():
    wolock = await GetSwitchbotDevices().get_locks()
    print(wolock)
    l = lock.SwitchbotLock(wolock['FF:02:53:A3:BB:11'].device,"13", "XXXXXXXXXXXXXXXXX","$")
    await l.get_basic_info()
    await l.unlock()
    
asyncio.run(main())
`
And the lock only "locks". 
I the lock is unlocked the lock locks whenever i send lock or unlock command but if the lock is locked the lock make a "beep" but nothing happens.
the result of get_basic_info is alwais = bytearray(b'\x05') = "Unsuccessful, please try again"
and the result of self._send_command(command) when lock or unlock is = bytearray(b'\x01\x82L\x00m\x00')  or bytearray(b'\x01\x82Z\x00m\x00') when do nothing and bytearray(b'\x01\x90Z\x00m\x00') when locks.

i have an european lock pro with the latest firmware.

chrismelba, 2024-06-10:
You can contact switchbot support directly if you want to change your main
account over to email. Not sure if this will help your problem though.

incre77, 2024-06-11:
very thanks @chrismelba. I did that and now i have my account correct. But i discovered what if i create a guest user in the app with an different email and i use that account to get the pass with the script i get the same ble-key as the main account.

Im trying to send unlock command from an esp32 after a nfc reading... i tried to port a part of the code to c++ but im stucked in the cipher part. Anyone knows something about this.I didnt found any related code for esp32 neither the non pro lock.

Anyway. thanks you very much to every body who made this locks working out of the web-api. :)

varyamereon, 2024-07-04:
Thanks for your hard work everyone! I haven't had a chance to test mine yet as I can't get it to authorise for some reason, but I noticed someone posted [in the HA repo](https://github.com/home-assistant/core/issues/121165) that night latch isn't working for them.

chrismelba, 2024-07-04:
Great to have it in HA! However my lock isn't showing state in HA if I manually change it? Ideally I'd like the lock to automatically lock at night or when i leave the house, however if I've opened it manually that won't work, as HA will think it's already locked

ollo69, 2024-07-04:
Just opened issue [#121254](https://github.com/home-assistant/core/issues/121254) in HA, seems to be the same issue descripted here.
Available for any debug to help to fix this, I have EU model with firmware V1.8 and battery pack

szclsya, 2024-07-05:
So turns out the lock pro indeed has a different passive mode data format (duh), and the old parser happened to work with the data I used for testing and align with the expected result when I wrote the first PR...

Anyways, after some trail and error I've written a new parser(szclsya/pySwitchbot@9d632bb). Currently I've only verified these to be correctly parsed:

+ battery
+ calibration
+ status
+ door_open
+ unclosed_alarm
+ unlocked_alarm
+ auto_lock_pause

So we will still need to verify these:

+ update_from_secondary_lock
+ double_lock_mode
+ night_latch

As a bonus, we can now determine if the lock is opened by hand by the `manual` field.

If anyone want to contribute you can run the following script (and grep `switchbot.adv_parsers.lock` as it's very verbose) to watch all events from the lock pro:

```python
#!/usr/bin/env python3

import switchbot
import asyncio
import logging

async def main():
    while True:
        getdevices = switchbot.discovery.GetSwitchbotDevices(1)
        await getdevices.discover()

if __name__ == '__main__':
    logging.basicConfig()
    logging.getLogger().setLevel(logging.DEBUG)
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
```
 Note that `mfr_data` will contain the MAC address of your lock, so remove the first 16 characters if you don't want to leak it. (shouldn't be a huge deal tho)

ollo69, 2024-07-05:
For me the parser from @szclsya branch (the one from his last commit) is working properly.
I tried to find information about battery pack in the returned raw data but I found nothing (nothing change removing and adding 1 battery pack at time).

Thanks!!!

chrismelba, 2024-07-06:
I've just updated ha to 2024.7.1

Now if I have the device open and manually trigger it I can see it change state, but if I don't have HA open and manually press the lock then ha hadn't updated after 5 minutes.

Does there need to be some kind of trigger to retrieve state or something?

I'm connecting via esphome Bluetooth gateway if that matters

ollo69, 2024-07-06:
> I've just updated ha to 2024.7.1
>
> Now if I have the device open and manually trigger it I can see it change state, but if I don't have HA open and manually press the lock then ha hadn't updated [...]

At this moment status for Lock Pro is not properly reported.
You must wait for new PR on this repository with the fix from @szclsya and for a new release of HA that include a PR that bump new version of this library in HA Switchbot integration.

mwolter805, 2024-07-11:
If possible, can someone help with adding the ability to see the charge level of both batteries in the rechargeable [Dual Power Pack](https://us.switch-bot.com/products/switchbot-lock-pro?variant=47258771521769)?

As noted here https://github.com/Danielhiversen/pySwitchbot/issues/233#issuecomment-2211182237 and https://github.com/Danielhiversen/pySwitchbot/pull/243#issuecomment-2221491972 the detailed battery info is indeed not sent in passive mode.

Please let me know if I can provide any info.

jensweimann, 2024-07-19:
Now that this feature is in the release of Home Assistant 2024.7.3 I tried it and at first, the night latch feature didn't show up. 

In the debug logs was this:
` [switchbot.adv_parsers.lock] {'battery': 84, 'calibration': True, 'status': <LockStatus.NOT_FULLY_LOCKED: 6>, 'door_open': False, 'update_from_secondary_lock': False, 'double_lock_mode': False, 'unclosed_alarm': False, 'unlocked_alarm': False, 'auto_lock_paused': False, 'night_latch': False}`
 
Then I locked and unlatched in the app and the night_latch flag became True. Then I tried it with home assistant a couple of times. Then I checked the log again and to my surprise the night_latch flag was false again.

**It seems that the flag gets disabled when unlatching trough home assistant**. What about this comment?

> @szclsya I have the EU lock and logged my bluetooth traffic. COMMAND_UNLOCK_WITHOUT_UNLATCH should be 0f4e01010001a0

In the code is this currently:  `SwitchbotModel.LOCK_PRO: f"{COMMAND_HEADER}0f4e01010000a0",`

Michal4K, 2024-07-19:
@jensweimann That nibble just indicate source of the action as described here - https://github.com/Danielhiversen/pySwitchbot/pull/241#issuecomment-2155066260.

schummar, 2024-07-22:
I have two locks, both with night latch. After the latest update one of them shows with the option to open, the other does not. Both bought together and same firmware version, so I would expect them to behave the same way.

What about owners of the non-night-latch version? Do some of you also see the "open" option? That might show that we have the wrong bit and it's indeed random chance. 😉

mwolter805, 2024-07-22:
@Michal4K I am not using the night latch feature.

Michal4K, 2024-07-22:
Then I think the night latch indicating bit was shifted up for few bits and the right one could be this one.
 
`"night_latch": bool(mfr_data[9] & 0b00010000)`

Michal4K, 2024-07-23:
@jsprotky 
For every one who would like to use nightlatch feature and they don't see it in Home Assistant now because of the issue with the adv parser. There is a way to manually overwrite entity feature set. 

By customizing entity in HA configuration file:
`
homeassistant:
  customize:
    lock.<<your_lock_entity>>:
      supported_features: 1
`
Or you can also change this in developer tool -> States, but it will last only until switchbot integration reload.

csmale, 2024-07-23:
> > @jsprotky
> > For everyone who would like to use nightlatch feature and they don't see it in Home Assistant now, because of the issue with the adv parser. There is a way to manually overwrite entity feature set.
> > By customizing entity in HA configuration file:
> > ```
> > homeassistant:
> >   customize:
> >     lock.<<your_lock_entity>>:
> >       supported_features: 1
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Or you can also change this in developer tool -> States, but it will last only until switchbot integration reload.
> 
> I get "entity does not support this service" when I change the supported_features attribute and try to unlatch the lock. The button to unlatch the lock is there but not working. Are you sure that pyswitchbot will unlatch just because HA sent the unlatch request?

I have the same. The UI includes the right buttons but a call to lock.open on lock.* gives "Failed to call service lock.open. Entity lock.lock_pro_xxxx does not support this service."

A call to lock.unlock is accepted but it goes too far and opens the door as well.

Michal4K, 2024-07-23:
You are right. It does not work. Only way to make it work will be change ADV parse to return always True for night latch feature.
But this require going inside your HA finding the switchbot library and it will be overwrite with every update.
Sorry for not working advice.

balu79, 2024-08-01:
I just changed my batteries (replaced normally batteries by battery pack) and seems that caused my previous night latch support to stop working (and supported_features was updated from 1 to now 0).

Pascal4Games, 2024-08-13:
> You are right. It does not work. Only way to make it work will be change ADV parse to return always True for night latch feature. But this require going inside your HA finding the switchbot library and it will be overwrite with every update. Sorry for not working advice.

Another workarround for the moment would be to download the switchbot integration and add it as a custom integration. Downsite of the ajustment is that when you have multiple locks and not all need unlatch it doesn't work and if the issue get fix  it doesn't auto update.

You do have to change to files to get it to work:
manifest.json
```json
add don't forgot the "," at the end of the last line.
"version": "1.0.8"
``` 
and in lock.py

```python
By "def __init__" after change looks like this
        """Initialize the entity."""
        super().__init__(coordinator)
        self._async_update_attrs()
        #if self._device.is_night_latch_enabled():
        self._attr_supported_features = LockEntityFeature.OPEN

and by "async def async_unlock" after change looks like this:
        """Unlock the lock."""
        #if self._device.is_night_latch_enabled():
        self._last_run_success = await self._device.unlock_without_unlatch()
        #else:
        #    self._last_run_success = await self._device.unlock()
        self.async_write_ha_state()
``` 

Hopefully this is useful.

Xyz00777, 2024-08-19:
@Pascal4Games do you have possibly time to do an little step by step guide because at least i am a little bit helpless how to do it..?
or is the fix near to be released?

balu79, 2024-08-20:
> @Pascal4Games do you have possibly time to do an little step by step guide because at least i am a little bit helpless how to do it..? or is the fix near to be released?

This is how I did it. Probably the step to download the switchbot integration is not the best, but I couldn't figure out another way.
1.) Went to the switchbot bluetooth integration on github (https://github.com/home-assistant/core/tree/dev/homeassistant/components/switchbot) and downloaded all the 14 python and json files manually one by one
2.) Modified the 2 above mentioned python and json files according to the intrductions from Pascal4Games
3.) Created a folder called "switchbot" on my PC to which I moved all 14 downloaded and modified files
4.) Uploaded that switchbot folder with the files to config\custom_components
5.) Restarted home assistant and checked that there are no errors in the logs of home assistant related to the switchbot (bluetooth) integration as well that within the integrations page the switchbot integration now shows a yellow icon for custom integration
6.) Test functionality and be happy ;-)

Pascal4Games, 2024-08-20:
Sorry, I hadn't seen your response yet, but I'm glad you managed to do it yourself.
To download the integration I used https://download-directory.github.io/
It's not a fix for everyone, but only for people who have the lock-pro, want to use the latch function, don't have a lock that doesn't need the latch function and don't want to wait until the problem is solved.

Michal4K, 2024-08-20:
I opened a PR in HA to which allow forcing night latch operation.
https://github.com/home-assistant/core/pull/124326

It simply adds checkbox to option configuration to force lock to operate as night latch was detected.
[image]

Pascal4Games, 2024-08-21:
> I opened a PR in HA to which allow forcing night latch operation. [home-assistant/core#124326](https://github.com/home-assistant/core/pull/124326)
> 
> It simply adds checkbox to option configuration which force lock to operate as night latch was detected. ![20240820_21h51m41s_grim](https://private-user-images.githubusercontent.com/12269903/359625042-dd101f13-c4da-4163-9dc6-3d0bd35ae188.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyNjQ1ODIsIm5iZiI6MTcyNDI2NDI4MiwicGF0aCI6Ii8xMjI2OTkwMy8zNTk2MjUwNDItZGQxMDFmMTMtYzRkYS00MTYzLTlkYzYtM2QwYmQzNWFlMTg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIxVDE4MTgwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBhZDYwZjY0MmFkNTkyNTFmYTM4ZGIyYzRiN2VjZjVlYjRjM2ZkZGM3YTI1NWFkMzEzM2YzNWZiZDlmMjFiODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XIoMHYN5_NiftQrcvn9tlBi5qVizJwavD4DQvAHmcoo)

That is indeed a better solution.
Thanks just tested and works.