sblibs/pySwitchbot #234: API 190 Error

Shirotaku, 2024-04-25:
Hello,

When running the get_encryption_key script, I get API 190 error. That's the same error that I get when logging into HAs Bluetooth integration. Any ideas?

`pySwitchbot-0.45.0>get_encryption_key.py MAC EMAIL
Password: 
Unexpected status code returned by SwitchBot API: 190`

Shirotaku, 2024-05-01:
@Danielhiversen do you think you can look into this? :)

NdK73, 2024-05-01:
Same here. Maybe it's related to changes in the API?

Jeffnl98, 2024-05-01:
Same for me. tried to make a new bluetooth proxy and moved it nearby but still the same error....

Shirotaku, 2024-05-02:
@NdK73 @Jeffnl98 Hey, I've opened a ticket in SwitchBot's API Repository (https://github.com/OpenWonderLabs/SwitchBotAPI/issues/296). Maybe give them a heads-up, too, so they can fix this issue asap.

Jaheiro, 2024-05-13:
Hi Folks, it seems that this 190 error is thrown in response to request in following method:  https://github.com/Danielhiversen/pySwitchbot/blob/343f6ccecb357245316aa3d7508372e8f8010006/switchbot/devices/lock.py#L90

If i understand correctly, code is not using Switchbot Api from @Shirotaku comment but rather tries to communicate with Switchbot servers with same request app would do. 

Unfortunately i don't have idea, how we could figure out what changed in their internal API, that throws error 190 now.
I tried to find a way we could get encryption key from their SwitchbotAPI instead, but with no luck, device info and device status is rather vague, especially if you don't have Switchbot Hub like me :(

micaelp, 2024-05-16:
Hi @Jaheiro the creator of the Switchbot MQTT service for HA (https://github.com/hsakoh) advised there was a new API URL however I am currently still getting error 190 (the example code he gave me elaborates on this saying I am not the device owner).  I've been back and forth with Switchbot for the last week and they removed my lock from all accounts I believe.  I've just re-registered and recalibrated and still get the same error!  @Shirotaku linked to the API repo chat above which may, or may not help?

alexschultze, 2024-05-16:
+1 one this issue. Initially error 190 through home assistant and also through pySwitchbot now.
I assumed it may be related to app login so i deleted my complete app account and readded it, with no success.

Jaheiro, 2024-05-16:
Howdy @micaelp, i'm guessing that along with that url change for API, something else had to change as well. 
I think problem is not in this api https://github.com/OpenWonderLabs/SwitchBotAPI as it is different url, but rather in internal communication that Switchbot created for their mobile application and both HA integration and pySwitchbot uses.

I suspect, they might have changed way that Switchbot application authorizes itself to get encryption keys of lock device, to send commands directly to the lock through bluetooth so we're sending request to proper place, but server can't confirm that we're actual owner of that specific lock.

alexschultze, 2024-05-17:
I am not sure who is in charge of that code (lock.py) which handles receiving the keys. But that was an excellent reverse engineering.

Without beeing an expert, I assume from the android apk that the Amazon Cognito pool has changed and the original authentication method is now not supported anymore. It may have been changed to "USER_SRP_AUTH" and "REFRESH_TOKEN_AUTH", whereas "USER_PASSWORD_AUTH" is disabled on the new pool.

See: [](https://docs.aws.amazon.com/cognito-user-identity-pools/latest/APIReference/API_InitiateAuth.html)

alexschultze, 2024-05-21:
The cognito pool has changed, and the new pool has SRP auth. I have tried implementing an SRP authentification into the new pool, but so far I could not get it to work (using my APP ID credentials). Maybe there is some caveat to the credentials to be used to identify to the AWS cognito pool. 

` "CognitoUserPool": { "Default": { "PoolId": "us-east-1_S5kbwuSkN", "AppClientId": "7o4edsh9v2glcmvu821m5c8pod", "AppClientSecret":"2bkcq702hj97gucjiricvl14hd3uqebifqimk93t6h04gkj2f4a", "Region": "us-east-1" }`

alexschultze, 2024-05-21:
So the assumption that the identification has changed was somewhat correct. The access_token is now requested from url="https://account.api.switchbot.net/account/api/v1/user/login", including additional information in the request (clientID ?? ).
`{ "clientId": "xxxxxxx",
 "username": "app user",
                "password": app pass",
                "grantType": "password",
                "verifyCode": ""}`

The following request to to "https://wonderlabs.eu.api.switchbot.net/wonder/keys/v1/communicate" is as usual with the new access_token.

**Workaround for key retrieval**
I finally only managed to get my keys using mtmproxy by observing the communication.

dsypniewski, 2024-05-22:
@alexschultze Thanks for tagging me, I missed this issue.

I've looked into this a little bit as well and at least for me just updating the PoolId to the new one (`us-east-1_S5kbwuSkN`) without changing the AppClientId and AppClientSecret seems to work fine without needing to change anything else in the authentication flow. But my account also worked fine with the previous PoolId so could others verify that it works for their (possibly newer) accounts as well?

dsypniewski, 2024-05-22:
@alexschultze also the authentication flow you posted unfortunately doesn't work for my account and returns the 190 error, so it would seem to me that there is some kind of divide in SwitchBot accounts that we'd need to figure out. One option that comes to mind is just try both auth methods and one should succeed.

Shirotaku, 2024-05-22:
@alexschultze your code worked perfectly for me, thanks!

hsakoh, 2024-05-22:
@dsypniewski 

I have implemented some additional ideas. (This is a [cross-post](https://github.com/OpenWonderLabs/SwitchBotAPI/issues/296#issuecomment-2125652922))
https://gist.github.com/hsakoh/fdeadc915b7c62ed443604ad3b364f2d

1. Retrieving the access token
   `https://account.api.switchbot.net/account/api/v1/user/login`
2. Determining the botRegion
   `https://account.api.switchbot.net/account/api/v1/user/userinfo`
3. Obtaining the key and keyId
   `https://wonderlabs.{botRegion}.api.switchbot.net/wonder/keys/v1/communicate`

Whether the botRegion obtained in Step 2's UserInfo is wonderlabs.{botRegion} will require verification by various users.
(At least for Japanese users, `ap` is returned, so it worked correctly.)

If these steps are implemented in PySwitchbot, it might make the setup process easier for everyone.

PR #236 (2024-05-23): Fix lock encryption key retrieval
It seems that SwitchBot has made some changes to their accounts and the previous key retrieval method doesn't work for everybody. This is based on @alexschultze and @hsakoh research, and verified myself with the SwitchBot Android application code.
Since cognito authentication is no longer used I also removed the boto3 dependency.

Fixes #234

Edit: I noticed only after creating this PR that there already is one addressing the exact same issue #235
closed 2024-05-23 by bdraco