sblibs/pySwitchbot #239: get_encryption_key is not working: Authentication failed: Incorrect username or password, status code: 2020

carferrer, 2024-05-27:
I can`t get the locks encryption key. I get the error "Authentication failed: Incorrect username or password, status code: 2020". I have double check user, password, change password and nit work. II can login to the website https://eu.switch-bot.com. My country region is Spain.

carferrer, 2024-05-30:
I have create a new account on switchbot website and works fine. It looks like have been and account update and with my old one (7 devices) does'nt work and with the new one works (1 device).

wunderbaum, 2024-07-05:
I had to type it manually, everywhere else it was possible with copypaste. Maybe this helps someone.

huyuwei1996, 2024-07-18:
The users of this website and the users of the SwitchBot app do not share the same account. If you have only logged in through the website, you are likely using an official website account. 
Can you enter and login with your username and password on the app?

nerumo, 2024-08-12:
I'm struggling with this. I even created a fresh account with a different email (registered within the app) and I get the 2020 error. My region is switzerland. Are there different Apis for different regions?

Which credentials do I need? from the website or from the app?

carferrer, 2024-08-12:
You need the credential used in the app.