sblibs/pySwitchbot #244: Timeout error while trying to authenticate

varyamereon, 2024-07-18:
I am receiving a timeout error while trying to authenticate my lock pro via home assistant. It’s the only SwitchBot device I have. Here is the log file from HA:

```
Logger: aiohttp.server
Source: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
First occurred: 16 July 2024 at 08:37:21 (5 occurrences)
Last logged: 09:15:19

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 214, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/dbus_fast/aio/message_bus.py", line 385, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 363, in establish_connection
    await client.connect(
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 310, in connect
    connected = await super().connect(**kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 368, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 415, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 518, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/switchbot/config_flow.py", line 198, in async_step_lock_auth
    return await self.async_step_lock_key(key_details)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/switchbot/config_flow.py", line 239, in async_step_lock_key
    if not await SwitchbotLock.verify_encryption_key(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 101, in verify_encryption_key
    lock_info = await lock.get_basic_info()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 257, in get_basic_info
    lock_raw_data = await self._get_lock_info()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 299, in _get_lock_info
    _data = await self._send_command(
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 356, in _send_command
    result = await self._ensure_encryption_initialized()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 371, in _ensure_encryption_initialized
    result = await self._send_command(
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 354, in _send_command
    return await super()._send_command(key[:2] + "000000" + key[2:], retry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 185, in _send_command
    return await self._send_command_locked(key, command)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 419, in _send_command_locked
    await self._ensure_connected()
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 279, in _ensure_connected
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 387, in establish_connection
    _raise_if_needed(name, device.address, exc)
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 327, in _raise_if_needed
    raise BleakNotFoundError(msg) from exc
bleak_retry_connector.BleakNotFoundError: E7-38-70-F8-C4-11 (E7:38:70:F8:C4:11) - E7:38:70:F8:C4:11: Failed to connect after 4 attempt(s): TimeoutError
```

This results in the process displaying an Unknown Error in HA and the device never gets added.

undauntedjogging, 2024-07-23:
I have the same issue like @varyamereon 

`Protokolldetails (FEHLER)
Logger: aiohttp.server
Quelle: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
Erstmals aufgetreten: 20:20:42 (3 Vorkommnisse)
Zuletzt protokolliert: 20:32:08

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 214, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/dbus_fast/aio/message_bus.py", line 385, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 363, in establish_connection
    await client.connect(
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 310, in connect
    connected = await super().connect(**kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 31 lines cut ...]
    _data = await self._send_command(
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 356, in _send_command
    result = await self._ensure_encryption_initialized()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 371, in _ensure_encryption_initialized
    result = await self._send_command(
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/lock.py", line 354, in _send_command
    return await super()._send_command(key[:2] + "000000" + key[2:], retry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 185, in _send_command
    return await self._send_command_locked(key, command)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 419, in _send_command_locked
    await self._ensure_connected()
  File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 279, in _ensure_connected
    client: BleakClientWithServiceCache = await establish_connection(
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 387, in establish_connection
    _raise_if_needed(name, device.address, exc)
  File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 327, in _raise_if_needed
    raise BleakNotFoundError(msg) from exc
bleak_retry_connector.BleakNotFoundError: D5-3E-63-87-40-3A (D5:3E:63:87:40:3A) - D5:3E:63:87:40:3A: Failed to connect after 6 attempt(s): TimeoutError
`

DJOma, 2024-08-22:
Hi,

I can not say, if this is the reason for your problems but for me it was the bluetooth dongle.
It was detected by HA fine and found the Lock Pro, but was never able to connect to it.
I even got the auth details via the script and it still timed out.

I then tried an ESP Home bluetooth proxy and it instantly worked.

varyamereon, 2024-08-22:
*Sigh* You might be right, thanks @DJOma. Worth a try anyway, I'll have to pick something up from Amazon. Any recommendations?

DJOma, 2024-08-22:
[https://esphome.io/projects/?type=bluetooth](url)

Thats all you need. It is really simple.
I was lucky and had an unused ESP32 laying around in a shelf.

bdraco, 2024-08-22:
If you want one that won’t drop out
https://a.aliexpress.com/_mK8dDw8

varyamereon, 2024-08-28:
Thanks for the suggestions, I did indeed pick up an ESP32 and set the lock up via bluetooth proxy and it works absolutely fine, so I would say that is the fix for anyone who is interested!

closed 2024-08-28 by varyamereon