sblibs/pySwitchbot #246: feature request: support for the Water Leak Detector

MikeDembek, 2024-08-17:
Hey, I've gotten a couple of the new Water Leak Detectors and would be interested in support for those.  It looks like there are a number of others in the HA community interested as well:

https://community.home-assistant.io/t/switchbot-bluetooth-water-leak-detector/749332

Here's the API documentation:

https://github.com/OpenWonderLabs/SwitchBotAPI?tab=readme-ov-file#water-leak-detector

Thanks for any help, and if there's more info I can provide I'll try to help out.

luc-ass, 2024-09-05:
Hey there. Great to see that there is more interest in this. I would love to help out, but can not figure out on where to start. @Danielhiversen could you point me in the right direction? I have a spare sensor laying around...

luc-ass, 2025-01-05:
@MikeDembek looks like this can be closed, as it has been merged

closed 2025-01-05 by bdraco