sblibs/pySwitchbot #248: Lock Pro

dackel1, 2024-09-06:
Since Update to HA Core 2024.9 the function to "pull the latch" (german: Falle ziehen) is missing. 
Before the update: On the dashboard it was possible to lock and unlock AND to open the door - this function was great. HA even asked on first tap: "really want to open". 
Since the update to 2024.9 the functions are limited to lock and (unlock&open). 
The "old" behaviour was much better, worked flawless and was much more safe - it was nearly impossibility to remote OPEN the door by accident. 
Please restore the old behaviour.

Michal4K, 2024-09-06:
Yeh, this library now has disabled nightlatch detection... on purpose as the detection was not working properly. There will be work around for this which will allow you to force nightlatch in HomeAssistant. Unfortunately this change didn't make it into the 2024.9...

Michal4K, 2024-09-06:
for now you can fix it by updating pySwitchbot lib to my version which has set nightlatch to be always enabled - https://github.com/Danielhiversen/pySwitchbot/pull/245#issuecomment-2251521762

dackel1, 2024-09-08:
@Michal4K Thank you for your quick reply and the workaround.
Do I understand you right and you want to restore the latch-function in 2024.10? 
As HA beginner I'm not shure how to get back from the workaroud to the release-channel, so waiting for 2024.10 would really be an option.

Michal4K, 2024-09-10:
Yes, you can wait for 2024.10. When it will be released and you update to it. You will need go to Settings -> Device and Services -> SwitchBot Bluetooth. Then hit configure on your lock entry... there will be checbox with text "Force Nightlatch operation mode".

dackel1, 2024-09-10:
@Michal4K 
This is great news. Thanks for this information, your work & help.

andreasbrett, 2024-11-23:
@dackel1 Since the integration works as expected now, could you close the issue?

dackel1, 2024-11-23:
@Michal4K 
Of course. I'm new to github, sorry

closed 2024-11-23 by dackel1