sblibs/pySwitchbot #284: SwitchBot devices require active scans to be discovered

bdraco, 2024-12-31:
The service data is only send in the scan response which makes it impossible for Home Assistant to discover these devices with passive only scans.

Typical solutions other vendors use:

- Send the service data in the base adv (sometimes a problem since its limits to 31 bytes). The usual trade off is to move the local name into the scan response only to make room
- Periodicity send the service data in the base adv, even if its only every 15 minutes or so.

luc-ass, 2025-01-05:
@bdraco is there something we can do here? Wouldn't that be something Switchbot needs to fix?

bdraco, 2025-01-05:
Its something that would need to be changed at the firmware level

bdraco, 2025-08-30:
I was thinking about the devices not being discoverable in passive mode, however I think one solution would be to login to the cloud service, download all the mac address and models in the config flow, see which ones are in range and than use that to discover all the devices in range.

PR #397 (2025-09-10): Fix device discovery when Bluetooth adapter is in passive scanning mode
Fixes #284

## Summary

This PR adds support for discovering Switchbot devices when the Bluetooth adapter is performing passive scans by fetching device information from the Switchbot cloud API and caching MAC address to model mappings.

## Problem

When a Bluetooth adapter is configured for passive scanning (not sending scan requests), it only receives passive advertisements from devices. These passive advertisements from Switchbot devices contain only manufacturer data without service data, making it impossible to identi [... cut]
closed 2025-09-10 by bdraco