sblibs/pySwitchbot #299: Not "all" Switchbot Meter Pro CO2 detected

marcelbarkh, 2025-02-09:
Hi,

I have several Switchbot Meter Pro CO2 device, but not all of them were detected in my HA.
After debugging it worked on my system for ALL device, if I added a "manufacturer_data_length" in adv_parser.py:

`    "5": {
        "modelName": SwitchbotModel.METER_PRO_C,
        "modelFriendlyName": "Meter",
        "func": process_wosensorth_c,
        "manufacturer_id": 2409,
        "manufacturer_data_length": 16,
    },`

I'm actually confused how it worked for the "working" device without that change..?

Cheers

bdraco, 2025-02-09:
Switchbot devices need active scans to be detected as model data is in the service data.

bdraco, 2025-02-09:
https://github.com/sblibs/pySwitchbot/issues/284

marcelbarkh, 2025-02-09:
Yes - I had active scan enabled. But even with this, not all devices were recognized...
As I said after debugging WITH ACTIVE SCAN ON I found out that in `_parse_data` the branch `if model_data.get("manufacturer_data_length") == len(_mfr_data):` failed for the device with the connection problem.
After adding the `manufacturer_data_length` it worked...
Without that, only some of the devices were recognized - regardless of why they were recognized...

bdraco, 2025-02-09:
It should only get there if the service data is missing since it should find it in 
```
    if _switchbot_model and _switchbot_model in _SWITCHBOT_MODEL_TO_CHAR:
        _model = _SWITCHBOT_MODEL_TO_CHAR[_switchbot_model]
```
first

marcelbarkh, 2025-02-09:
In fact - in the switchbot HA component in `config_flow.py` in `_async_discover_devices`, which is called in `async_step_user`, the `advertisement_data` is `AdvertisementData(manufacturer_data={2409: b'\xb0\xe9\xfeUJ\xae\x12\xe4\x00\x95\xa6\x00\x04\x02\xfb\x00'}, tx_power=-127, rssi=-62)`
=> it has no `service_data`
And right now I don't know how to trigger `async_step_bluetooth`, where `_async_discover_devices` is also called...