sblibs/pySwitchbot #311: README is not telling how to send a BLE command

dr-ni, 2025-03-19:
Please add a description or some examples (for switchbot, lock pro, ...) to completeley show the process for a cloudless usage of switchbot devices.
This repo is usless with no documentation

dr-ni, 2025-03-21:
Is this the correct usage?

```
import asyncio
from switchbot.discovery import GetSwitchbotDevices
from switchbot.devices import lock

async def main():
    wolock = await GetSwitchbotDevices().get_locks()
    print()
    print(wolock)
    print()
    await lock.SwitchbotLock(wolock['"<ble-mac>").'].device, "<key-id>", "<encryption-key>").lock()
    await lock.SwitchbotLock(wolock['"<ble-mac>").'].device, "<key-id>", "<encryption-key>").unlock()

asyncio.run(main())
```

dr-ni, 2025-03-22:
The above unlock command is also locking my lock pro?
So I cannot unlock. Is this a bug?

See also #312

dr-ni, 2025-03-24:
some more news:

I could get it working after changing lock.py like this
```python
COMMAND_HEADER = "57"
COMMAND_LOCK_INFO = {
    SwitchbotModel.LOCK: f"{COMMAND_HEADER}0f4f8101",
    SwitchbotModel.LOCK_PRO: f"{COMMAND_HEADER}0f4f8102",
}
COMMAND_UNLOCK = {
#    SwitchbotModel.LOCK:     f"{COMMAND_HEADER}0f4e01011080",
    SwitchbotModel.LOCK: f"{COMMAND_HEADER}0f4e0101000080",
}
COMMAND_UNLOCK_WITHOUT_UNLATCH = {
#    SwitchbotModel.LOCK: f"{COMMAND_HEADER}0f4e010110a0",
#    SwitchbotModel.LOCK_PRO: f"{COMMAND_HEADER}0f4e01010000a0",
}
COMMAND_LOCK = {
#    SwitchbotModel.LOCK: f"{COMMAND_HEADER}0f4e01011000",
    SwitchbotModel.LOCK: f"{COMMAND_HEADER}0f4e0101000000",
}
```

and got the following output:

```shell
uwe@rp5:~/pySwitchbot $ bin/python3 test_lock.py
# Found Lock:
{'XX:XX:XX:XX:XX:XX': SwitchBotAdvertisement(address='XX:XX:XX:XX:XX:XX', data={'rawAdvData': b'$\x80a', 'data': {'battery': 97, 'calibration': True, 'status': <LockStatus.UNLOCKED: 1>, 'door_open': True, 'update_from_secondary_lock': False, 'double_lock_mode': False, 'unclosed_alarm': False, 'unlocked_alarm': False, 'auto_lock_paused': True, 'night_latch': False}, 'model': '$', 'isEncrypted': False, 'modelFriendlyName': 'Lock Pro', 'modelName': <SwitchbotModel.LOCK_PRO: 'WoLockPro'>}, device=BLEDevice(XX:XX:XX:XX:XX:XX, Lock Pro), rssi=-38, active=True)}

uwe@rp5:~/pySwitchbot $ bin/python3 lock.py
# Found Lock:
{'XX:XX:XX:XX:XX:XX': SwitchBotAdvertisement(address='XX:XX:XX:XX:XX:XX', data={'rawAdvData': b'$\x80a', 'data': {'battery': 97, 'calibration': True, 'status': <LockStatus.UNLOCKED: 1>, 'door_open': True, 'update_from_secondary_lock': False, 'double_lock_mode': False, 'unclosed_alarm': False, 'unlocked_alarm': False, 'auto_lock_paused': True, 'night_latch': False}, 'model': '$', 'isEncrypted': False, 'modelFriendlyName': 'Lock Pro', 'modelName': <SwitchbotModel.LOCK_PRO: 'WoLockPro'>}, device=BLEDevice(XX:XX:XX:XX:XX:XX, Lock Pro), rssi=-31, active=True)}

/home/uwe/pySwitchbot/switchbot/devices/device.py:292: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  return self._device.rssi
/home/uwe/pySwitchbot/switchbot/devices/device.py:292: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  return self._device.rssi
Unsuccessful, please try again
No advertisement data to update
NoneType: None
uwe@rp5:~/pySwitchbot $ bin/python3 unlock.py
# Found Lock:
{'XX:XX:XX:XX:XX:XX': SwitchBotAdvertisement(address='XX:XX:XX:XX:XX:XX', data={'rawAdvData': b'$\x80a', 'data': {'battery': 97, 'calibration': True, 'status': <LockStatus.LOCKED: 0>, 'door_open': True, 'update_from_secondary_lock': False, 'double_lock_mode': False, 'unclosed_alarm': False, 'unlocked_alarm': False, 'auto_lock_paused': True, 'night_latch': False}, 'model': '$', 'isEncrypted': False, 'modelFriendlyName': 'Lock Pro', 'modelName': <SwitchbotModel.LOCK_PRO: 'WoLockPro'>}, device=BLEDevice(XX:XX:XX:XX:XX:XX, Lock Pro), rssi=-28, active=True)}

/home/uwe/pySwitchbot/switchbot/devices/device.py:292: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  return self._device.rssi
/home/uwe/pySwitchbot/switchbot/devices/device.py:292: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  return self._device.rssi
Unsuccessful, please try again
No advertisement data to update
NoneType: None
uwe@rp5:~/pySwitchbot $ bin/python3 test_lock.py
# Found Lock:
{'XX:XX:XX:XX:XX:XX': SwitchBotAdvertisement(address='XX:XX:XX:XX:XX:XX', data={'rawAdvData': b'$\x80a', 'data': {'battery': 97, 'calibration': True, 'status': <LockStatus.UNLOCKED: 1>, 'door_open': True, 'update_from_secondary_lock': False, 'double_lock_mode': False, 'unclosed_alarm': False, 'unlocked_alarm': False, 'auto_lock_paused': True, 'night_latch': False}, 'model': '$', 'isEncrypted': False, 'modelFriendlyName': 'Lock Pro', 'modelName': <SwitchbotModel.LOCK_PRO: 'WoLockPro'>}, device=BLEDevice(XX:XX:XX:XX:XX:XX, Lock Pro), rssi=-31, active=True)}
```

dr-ni, 2025-03-24:
This shows me that instead of `SwitchbotModel.LOCK_PRO` the `SwitchbotModel.LOCK` is used - althogh that `wolock` has the correct `'modelName': <SwitchbotModel.LOCK_PRO` of my Lock-Pro.  And also `wolock`  has `'night_latch': False` but  `COMMAND_UNLOCK` is used instead of `COMMAND_UNLOCK_WITHOUT_UNLATCH`.

So i guess I have to send my commands in a different way, isn't it?

Michal4K, 2025-03-25:
Did you import it?
`from switchbot.const import SwitchbotModel`

dr-ni, 2025-03-25:
ok this import was missing

dr-ni, 2025-03-30:
see #312

closed 2025-03-30 by dr-ni