sblibs/pySwitchbot #353: Lock status abnormal in HA

zerzhang, 2025-06-09:
@bdraco Hi, Could you help to have a look for case https://github.com/home-assistant/core/issues/142316 and https://github.com/home-assistant/core/issues/146364 and give me some suggestion,  The lock works well but but the status is abnormal until the user reload the device

zerzhang, 2025-06-09:
2025-06-06 20:32:25.591 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Connected; RSSI: -73
2025-06-06 20:32:25.591 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Starting notify and disconnect timer; RSSI: -73
2025-06-06 20:32:25.592 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Subscribe to notifications; RSSI: -73
2025-06-06 20:32:25.613 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 570000000f210334
2025-06-06 20:32:25.729 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 010000002eb4b614affc1aee7d710a9c06f60f1c
2025-06-06 20:32:25.729 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb4176af819107394
2025-06-06 20:32:25.729 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:25.729 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb4176af819107394
2025-06-06 20:32:25.891 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb49011b12d10
2025-06-06 20:32:25.892 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb41a
2025-06-06 20:32:25.892 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:25.892 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb41a
2025-06-06 20:32:25.925 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb4176af819107394
2025-06-06 20:32:25.925 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Operation already in progress, waiting for it to complete; RSSI: -73
2025-06-06 20:32:25.983 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb47c33e2181073
2025-06-06 20:32:25.984 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Fire callbacks
2025-06-06 20:32:25.984 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:25.985 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb4176af819107394
2025-06-06 20:32:26.096 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb49011b12d10
2025-06-06 20:32:26.096 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb41a
2025-06-06 20:32:26.096 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:26.096 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb41a
2025-06-06 20:32:26.105 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb4176af819107394
2025-06-06 20:32:26.106 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Operation already in progress, waiting for it to complete; RSSI: -73
2025-06-06 20:32:26.232 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb47c33e2181073
2025-06-06 20:32:26.232 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Fire callbacks
2025-06-06 20:32:26.232 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:26.232 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb4176af819107394
2025-06-06 20:32:26.267 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb4176af819107394
2025-06-06 20:32:26.267 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Operation already in progress, waiting for it to complete; RSSI: -73
2025-06-06 20:32:26.320 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb49011b12d10
2025-06-06 20:32:26.320 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb41a
2025-06-06 20:32:26.320 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Already connected before obtaining lock, resetting timer; RSSI: -73
2025-06-06 20:32:26.320 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Sending command: 57342eb4176af819107394
2025-06-06 20:32:26.449 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb4176af819107394
2025-06-06 20:32:26.449 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Operation already in progress, waiting for it to complete; RSSI: -73
2025-06-06 20:32:26.536 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb49011b12d10
2025-06-06 20:32:26.536 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Scheduling command 57342eb41a
[... cut ...]
2025-06-06 20:32:27.143 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Notification received: 01342eb47c33e2181073
2025-06-06 20:32:27.143 DEBUG (MainThread) [switchbot.devices.device] None (CD:48:23:B9:72:94): Fire callbacks
2025-06-06 20:32:35.037 DEBUG (MainThread) [switchbot.adv_parsers.lock] mfr_data: cd4823b97294108a00640000
2025-06-06 20:32:35.037 DEBUG (MainThread) [switchbot.adv_parsers.lock] data: 248064

And the log seems to stop record after 27s until the device is reloaded, I want to confirm whether frequent updates may cause this problem

zerzhang, 2025-06-09:
[image]
And the log seems to stop record after 27s until the device is reloaded, I want to confirm whether frequent updates may cause this problem

bdraco, 2025-06-09:
At least from the one users log there looks to be an error in the encryption implementation: https://github.com/home-assistant/core/issues/142316#issuecomment-2954519931

zerzhang, 2025-06-09:
```
2025-06-09 14:35:34.241 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Fire callbacks
2025-06-09 14:35:42.650 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Executing timed disconnect after timeout of 8.5
2025-06-09 14:35:42.651 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Executing disconnect
2025-06-09 14:35:42.651 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Executing disconnect with lock
2025-06-09 14:35:42.651 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Disconnecting
2025-06-09 14:35:43.248 INFO (MainThread) [homeassistant.components.switchbot.light] Turning on light {'rgb_color': (255, 125, 51)}, address A0:85:E3:65:2C:06
2025-06-09 14:35:43.248 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Scheduling command 57bc0933b6c53ac59c5d6f5b
2025-06-09 14:35:43.248 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Connection already in progress, waiting for it to complete; RSSI: -127
4444: iv cleared for WoFloorLamp (A0:85:E3:65:2C:06) at A0:85:E3:65:2C:06
2025-06-09 14:35:44.993 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Disconnected from device; RSSI: -127
2025-06-09 14:35:44.994 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Disconnect completed successfully
2025-06-09 14:35:44.994 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Connecting; RSSI: -127
2025-06-09 14:35:44.994 INFO (MainThread) [habluetooth.wrappers] A0:85:E3:65:2C:06 - WoFloorLamp: Found 1 connection path(s), preferred order: hci1 (80:C8:AC:00:00:27) (RSSI=-44) (failures=0) (in_progress=0) (score=-44.0)
2025-06-09 14:35:51.278 DEBUG (MainThread) [homeassistant.components.switchbot.coordinator] A0:85:E3:65:2C:06: Switchbot data: {'rawAdvData': b'\x00\x00\x00\x00\x10\xd0\xb0', 'data': {'sequence_number': 44, 'isOn': True, 'brightness': 34, 'delay': False, 'network_state': 2, 'color_mode': 2, 'cw': 5363, 'r': 66, 'g': 100, 'b': 255, 'firmware': 0.9}, 'model': b'\x00\x10\xd0\xb0', 'isEncrypted': False, 'modelFriendlyName': 'Floor Lamp', 'modelName': <SwitchbotModel.FLOOR_LAMP: 'Floor Lamp'>}
2025-06-09 14:35:51.279 INFO (MainThread) [homeassistant.components.switchbot.coordinator] Device Floor Lamp 2C06 is online
2025-06-09 14:35:51.279 DEBUG (MainThread) [switchbot.devices.base_light] WoFloorLamp (A0:85:E3:65:2C:06): update advertisement: SwitchBotAdvertisement(address='A0:85:E3:65:2C:06', data={'rawAdvData': b'\x00\x00\x00\x00\x10\xd0\xb0', 'data': {'sequence_number': 48, 'isOn': True, 'brightness': 34, 'delay': False, 'network_state': 2, 'color_mode': 2, 'cw': 5363}, 'model': b'\x00\x10\xd0\xb0', 'isEncrypted': False, 'modelFriendlyName': 'Floor Lamp', 'modelName': <SwitchbotModel.FLOOR_LAMP: 'Floor Lamp'>}, device=BLEDevice(A0:85:E3:65:2C:06, WoFloorLamp), rssi=-43, active=True) (seq before: 44) (seq after: 48)
2025-06-09 14:35:51.280 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Scheduling command 570000000f2103bc
2025-06-09 14:35:51.280 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Operation already in progress, waiting for it to complete; RSSI: -127
2025-06-09 14:35:52.141 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Connected; RSSI: -127
2025-06-09 14:35:52.141 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Starting notify and disconnect timer; RSSI: -127
2025-06-09 14:35:52.142 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Subscribe to notifications; RSSI: -127
2025-06-09 14:35:52.503 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Sending command: 57bc0933b6c53ac59c5d6f5b
3333: iv: None,address: A0:85:E3:65:2C:06
5555: decrypting  iv None,address: A0:85:E3:65:2C:06
2025-06-09 14:35:52.693 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Notification received: 01bc093339aec4aa8da2069b322a
2025-06-09 14:35:52.694 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Already connected before obtaining lock, resetting timer; RSSI: -127
2025-06-09 14:35:52.694 DEBUG (MainThread) [switchbot.devices.device] WoFloorLamp (A0:85:E3:65:2C:06): Sending command: 570000000f2103bc
2025-06-09 14:35:52.694 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [123661607074080] Unexpected exception
Traceback (most recent call last):
  File "/home/zero/workspace/core/venv/lib/python3.13/site-packages/cryptography/utils.py", line 50, in _check_byteslike
    memoryview(value)
    ~~~~~~~~~~^^^^^^^
TypeError: memoryview: a bytes-like object is required, not 'NoneType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zero/workspace/core/homeassistant/components/websocket_api/commands.py", line 250, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/home/zero/workspace/core/homeassistant/core.py", line 2802, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/home/zero/workspace/core/homeassistant/core.py", line 2845, in _execute_service
[... cut ...]
  File "/home/zero/workspace/core/venv/lib/python3.13/site-packages/cryptography/hazmat/primitives/ciphers/modes.py", line 204, in __init__
    utils._check_byteslike("nonce", nonce)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/home/zero/workspace/core/venv/lib/python3.13/site-packages/cryptography/utils.py", line 52, in _check_byteslike
    raise TypeError(f"{name} must be bytes-like")
TypeError: nonce must be bytes-like
```
The iv value is cleared. This happens when you perform an action while the device is disconnecting.

zerzhang, 2025-06-20:
@bdraco  This problem seems to still exist, Should this exception be thrown to the top layer and then caught with HA

[image]

zerzhang, 2025-07-03:
@bdraco We send a command `COMMAND_ENABLE_NOTIFICATIONS` for lock series, this will cause the device to continue to report messages periodically. Is it possible that the device just reported a message before the disconnection, and the iv was cleared after the disconnection, causing the decryption failure? I'm not sure about this because it's a bit hard for me to reproduce.

[image]

bdraco, 2025-07-03:
Does this only happen with the lock pro?

zerzhang, 2025-07-03:
For all lock series

zerzhang, 2025-07-03:
Thanks, i will try to reproduce this as well!

bdraco, 2025-07-03:
It might be bluetooth stack dependent.   Assuming its the issue you describe (and I think it is), https://github.com/sblibs/pySwitchbot/pull/368 should fix it

bdraco, 2025-07-03:
The issue doesn't reproduce for me. I forgot I had already tried to test this.

zerzhang, 2025-07-03:
> It might be bluetooth stack dependent.   Assuming its the issue you describe (and I think it is), https://github.com/sblibs/pySwitchbot/pull/368 should fix it
 ok. i will have a try

bdraco, 2025-07-03:
I'm finishing up for the day. If you are happy with the fix in https://github.com/sblibs/pySwitchbot/pull/368 feel free to approve and merge it.

zerzhang, 2025-07-04:
@bdraco I reproduced this error and found that it was just as I had guessed.

[image]

Samywamy10, 2025-07-04:
@zerzhang in that screenshot where you're logging the `bytearray` can you log the decoded version as well? Wondering if you're also getting the stuck `LOCKING` status (instead of `LOCKED` or `UNLOCKED`) after issuing a lock/unlock command

zerzhang, 2025-07-04:
@Samywamy10 Sure, The first data is the data reported periodically by the lock when you lock or unlock it. and the bytearray data is the IV,  accroding to the log you can see when the device is disconnecting,  the IV was cleared and this lead to the decryption failure and raise RuntimeError,  After fix this, the lock's status should be normal